Monash-Connected-Autonomous-Vehicle/autoware #10: control

dylan-gonzalez, 2023-11-26:
Understand [Autoware Control](https://autowarefoundation.github.io/autoware.universe/main/control/autonomous_emergency_braking/)

* configure launch file
* ensure test cases are working

dylan-gonzalez, 2023-11-29:
Hey @yentung-yen @klo40 do you guys wanna schedule a call to go over this? Or do you understand

yentung-yen, 2023-11-29:
hey @dylan-gonzalez , yep could we schedule a call? @klo40 and I will be coming in on Friday too. Just not sure what time yet.

dylan-gonzalez, 2023-11-29:
We can meet on Friday, I'll be there all day so let me know a time that's suitable for both of you

dylan-gonzalez, 2023-11-29:
Also pinging @jytKelly - let us know if you're free Friday

jytKelly, 2023-11-29:
@dylan-gonzalez  I have work until 5 pm on Friday. I can come into the workshop to work on this task on Sunday for the whole day.

dylan-gonzalez, 2023-12-01:
Updated the docs for this issue

yentung-yen, 2023-12-01:
Following done on the beauty (1st dec):
- downloaded autoware ( located at ~/home/mcav/autoware)
- downloaded awsim. Could be run with:

`chmod +x awsim/AWSIM_v1.1.0\ \(1\)/AWSIM_v1.1.0/AWSIM_demo.x86_64`
`awsim/AWSIM_v1.1.0\ \(1\)/AWSIM_v1.1.0/AWSIM_demo.x86_64`

- downloaded rviz but could not run it. Following error outputed:

[image]

klo40, 2023-12-01:
01/12/2023 Kendrick and Yen:
Notes: Control launch:
-At first glance, seems to have all the parameters and configurations set up.
-For previous error: ROS may not be running, should do ros2 topic list to check what's running.

yentung-yen, 2023-12-05:
05/12/2023 Yen and Kelly:

- Couldn't run things on the beauty. Used the beast instead.
- Followed instructions on the [wiki](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware/wiki/Autoware#docker-installation), but map still did show up on rviz

[image]

- tried changing to using /maps folder by running this on the terminal instead. but map still did not show up:
`ros2 launch autoware_launch e2e_simulator.launch.xml vehicle_model:=sample_vehicle sensor_model:=awsim_sensor_kit map_path:=/home/mcav/maps/autoware_map/sample-map-planning/`

- also tried this:

[image]
but got this error:

[image]

I'll probably join Jackson and Kehan's working session this thursday to see how they ran it.

dylan-gonzalez, 2023-12-05:
The maps directory should be /home/mcav/maps/awsim_map if using AWSIM

dylan-gonzalez, 2023-12-05:
Also it doesn't look like `source /opt/ros/humble/setup.bash` worked because you are trying to execute it on the same line as `cd autoware`. Seems that Quang made a typo in his instructions.

I have updated the [wiki](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware/wiki/Autoware) with more detailed instructions. Please make sure to correctly set `<map_path>` as I have stated in the instructions. And make sure to first enter the docker container.