Monash-Connected-Autonomous-Vehicle/autoware #11: env200 urdf

dylan-gonzalez, 2023-11-26:
Create a URDF model for the StreetDrone Nissan ENV200

For reference there is a StreetDrone Twizy model [here](https://github.com/streetdrone-home/SD-TwizyModel)
- see if there is an online CAD model we can use to convert to URDF
- contact StreetDrone to see if they have a URDF model

x1 person

suey147, 2023-12-02:
I've come across a few CAD models of the Nissan e-NV200, but it seems the pricing is not suitable for our current needs. 
https://sketchfab.com/3d-models/nissan-e-nv200-van-2022-db60a9a208e846a09c849f4411739cbb 
https://3dmodels.org/3d-models/nissan-e-nv200-van-2014/ 
https://www.cgtrader.com/3d-models/car/suv/nissan-e-nv200-van-2014 

also Found a free 3d obj of  Nissan nv 200 dhl van, which looks similar to an env200. could use this if we can't find other CAD files to convert.
https://grabcad.com/library/nissan-nv200-dhl-van-1

dylan-gonzalez, 2023-12-03:
The last one seems promising (unless Streetdrone doesn't have one already)

dylan-gonzalez, 2023-12-04:
> Have we contacted or found out if Streetdrone has a URDF model already? I see that checkbox has been marked off

I believe @suey147 has made a post on the zendesk. Here is a [guide](https://www.notion.so/monashcav/StreetDrone-Help-Desk-74474274c31b46308efdb8b4dc5b5d56) on how to access the zendesk.

> I don't have much experience on URDF but I'm studying up on the link of more info on URDF and will prepare to work on this from tomorrow onwards

It might be good if you and @suey147 organise a working session to create a URDF from this model: https://grabcad.com/library/nissan-nv200-dhl-van-1

adar0024, 2023-12-05:
[image]
I have accessed the zendesk but I'm not sure the post is here?

adar0024, 2023-12-05:
Suey did make the post but it was made on a different account. We posted it on the MCAV account now.

We are now working on creating a URDF of aforementioned model

adar0024, 2023-12-08:
https://www.cgtrader.com/free-3d-models/car/car/nissan-e-nv200

adar0024, 2023-12-10:
@suey147 

StreetDrone has replied to us with an attachment of files that include the e-NV200 URDF model! No need to convert anything to URDF anymore :)

[image]

dylan-gonzalez, 2023-12-11:
Closing this issue ^

closed 2023-12-11 by dylan-gonzalez