Monash-Connected-Autonomous-Vehicle/autoware #13: Autoware Launching

dylan-gonzalez, 2023-11-26:
1 person?

- configure all the launch files for our setups for CARLA, AWSIM, Twizy, ENV200
- configure parameters

https://github.com/tier4/aip_launcher/blob/tier4/universe/common_sensor_launch/launch/ars408.launch.xml

AnthonyZhOon, 2023-11-30:
@Robin-Chea If you're on campus to work on this Fri I can drop in and help/discuss what we need to do

Robin-Chea, 2023-11-30:
> @Robin-Chea If you're on campus to work on this Fri I can drop in and help/discuss what we need to do

ok ill be in from 12-5 on friday.

Robin-Chea, 2023-12-04:
Tried to create the launch files for twizy however after using the current stuff we had we were missing some stuff eg config files that describe the links to each sensor. So instead i have created the basic struture to create these config files.
I have basically copied the sample description and launch file and configured it to the twizy sensors eg 8 camera, 1 imu, 1 lidar (VLP-16) and 1 gnss. 
I am following the guide https://autowarefoundation.github.io/autoware-documentation/pr-491/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/

However not sure if i should include the ultrasonic sensors. 

I skipped this part for now  
https://autowarefoundation.github.io/autoware-documentation/pr-491/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/creating-individual-params/

i skipped configuring the sensor launch for now and will do it later 
https://autowarefoundation.github.io/autoware-documentation/pr-491/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/creating-sensor-model/#sensor-launch.

For vehicle launch desciption the config files need to be filled out.

For vehicle mesh a dae or a fbx needs to be made.

However in https://github.com/Monash-Connected-Autonomous-Vehicle/mcav_autonomy/tree/main/visualisation/vehicle_model/meshes they were able to use the stl file. will try to attempt to use this.

Robin-Chea, 2023-12-05:
did the same but for the env200

AbBaSaMo, 2023-12-09:
Would it be okay if I change the [repo URLs](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware/blob/yen-kendrick-control/autoware.repos) in our Autoware fork to our other forks where relevant or would there be more to it than just that? 

Is this being worked on right now as part of this issue and do I need to wait for it if I want to test out changes to sensing launch in a new branch as part of the sensing issue?

Robin-Chea, 2023-12-09:
@AbBaSaMo Yes it should be okay to change the repo url to relevant forks, especially if it helps streamline our development process. Currently i am working on the config files for the Twizy and once its completed, ill update it in the branch for integration.

dylan-gonzalez, 2024-01-08:
@Robin-Chea @AbBaSaMo to finish this off this week

Robin-Chea, 2024-01-22:
The model is now working on autoware : 

[image]

[image]

Still need to add the ultra sensors and the gps/imu. Need to also double check the drivers for camera and velodyne