Monash-Connected-Autonomous-Vehicle/autoware #6: mapping

dylan-gonzalez, 2023-11-26:
Understand [Autoware Mapping](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-maps/)

- make a lanelet map of Monash Research Way
- how to make a pointcloud map of Monash Research Way
- ensure the Autoware map blocks are outputting the correct information - see [AVP Diagram](https://app.diagrams.net/#G1T9vIuZJyE1f5AhWn4aL91sWi9EbfGOO_)
- create test cases (if it makes sense to)

zcdai, 2023-11-28:
OSM is not suitable for our pointcloud requirements, the data they provide is low-fidelity. [This](https://learnosm.org/en/osm-data/file-formats/) highlights how they have elevation data and vector points to form the surface of an area, which aren't suitable for our 3D use-case.

This [tool](https://tools.tier4.jp/vector_map_builder_ll2/) helps us make lanelet2 files (vector maps), provided with a PCD map as a baseline. More details will have to be looked into via the documentation [here](https://docs.web.auto/en/user-manuals/)

AnthonyZhOon, 2023-12-11:
So, right now our mapping isn't very detailed so it's hard to create a vector map.
We need to consider how to record more detailed pointcloud maps

ZileiChen, 2023-12-13:
A few screenshots of our pcd map

[image]

This one is okay, there might be a few misalignment issues where you can see faint outlines of trees inside the parking lot. Its possible to make a vector map out of this since I am able to see the where the curbs are meant to be.

[image]

This one is a lot better, and was the last record we did, similar problems as the one above but the image quality is a lot higher compared to before.
Another way of trying to lidar scan we could try is to just drive back and forth in a straight line to minimize misalignments.

Attempting to create vector map now.

ZileiChen, 2023-12-17:
Had issues saving the pcd files. Used the Woodside car park scans from last semester to create a lanelet2 vector map instead for now.

[image]

zcdai, 2024-01-06:
Looking at [this](https://github.com/autowarefoundation/autoware_common/blob/main/tmp/lanelet2_extension/docs/turn_direction.png) tells us that if we want to make lots of parking spots, we need to break the long road into multiple small components.

[image]

[image]

After fiddling with it I got this:

[image]

Despite it looking very innocent, making a parking block involves:
- Creating a point on the center linestring
- Connecting each point to the corners on each parking spot via a linestring
    - You can kind of cheat the curved line by copying the first linestring and then manually modifying the end point ids, this is 
    still tedious however 
- Create a new linestring between the corners of the parking spot that intersect with the road
- Create a new linestring between the corners of parking spot that intersect with another parking spot
- Select all 4 linestrings that make a parking spot, and create an area

This tedious process makes scalability look very unlikely unless we learn of some shortcuts in the future.

Some other points:
Area subtypes may allow us to customise software logic when handling these zones in the future, for now they have been marked with 'parking'.

Creating many linestrings on top of each other may not be a visible problem to a map designer, but the software may treat it very differently and run into logic errors. 

Currently planning to cease work on woodside pcd. Instead, reduce the problem down into a simple 1-way road with only 1 parking spot. Once the single parking spot problem has been solved we can expand back into real-world usage.

[image]

Also how did tierIV make such clean curves...

[image]

zcdai, 2024-01-14:
Things looked at this week:

- osm simplified to the 1 lane case

- create map dir in the beast, but osm will not load, perhaps something to do with yaml files, 
    - noticed osm versions were different compared to sample files, and that no llonglat coordinates were saved, only local x y

- used their sample parking demo to see the features and implementation instead.

Their implementation of parking is very novel, and I'm not entirely sure how it works

This is what it looks like in rviz:

[image]

This is what it looks like in the map builder:

[image]

This makes scaling up much simpler, we simply have to cover the entire area with a polygon with a parking_lot subtype and insert linestrings with parking_space subtypes. It's unnecessary to create the arduous bends using this method.

Will look into why these implementation returns clean blocks in rviz and transfer this info into woodside parking

dylan-gonzalez, 2024-01-15:
How does autoware handle these zones ?
Specify parking x-y coords with pose

This week: parking into designated zones (might have to write our own custom node)

ZileiChen, 2024-02-11:
https://gitlab.com/autowarefoundation/autoware.auto/AutowareAuto/-/issues/678

A previous project was similar to what we had, in terms of mapping we designate each parking spot as an area. 

[image]

[image]

ZileiChen, 2024-02-11:
Another strategy used here is to 'narrow' the driving area such that instead of dividing the road up into segments to create parking areas, a second line is used to divide parking spots. This could also just be how the car was designed. There wasn't any turning arcs in these maps for each spot which means we can use autoware to somehow create a way to park into these spots without following the map

[image]

ZileiChen, 2024-09-09:
[woodside_lanelet_v3.zip](https://github.com/user-attachments/files/16926388/woodside_lanelet_v3.zip)

Updated map to have smoother z-level

ZileiChen, 2024-09-09:
[woodside_lanelet_v4.zip](https://github.com/user-attachments/files/16927284/woodside_lanelet_v4.zip)
this version is the full carpark (extrapolated the missing parts by eye)