Monash-Connected-Autonomous-Vehicle/autoware #7: sensing

dylan-gonzalez, 2023-11-26:
Understand [Autoware Sensing](https://autowarefoundation.github.io/autoware.universe/main/sensing/gnss_poser/)

make sure all sensor drivers on the Twizy & ENV200 are outputting the correct messages - see [AVP Diagram](https://app.diagrams.net/#G1T9vIuZJyE1f5AhWn4aL91sWi9EbfGOO_)

- research the difference between using [RadarTracks](https://github.com/ros-perception/radar_msgs/blob/ros2/msg/RadarTrack.msg) v Pointcloud
    - for reference our radar driver for the ENV200 is [smartmicro](https://github.com/Monash-Connected-Autonomous-Vehicle/smartmicro_ros2_radars)a, which outputs a pointcloud
- create test cases for the CI/CD pipeline
- launch file configurations for ENV200 & Twizy

AbBaSaMo, 2023-12-05:
@dylan-gonzalez any docs or people we should refer to when testing sensor drivers on the cars?

dylan-gonzalez, 2023-12-05:
for the van: me, @hubdwoo, @Jiawei-Liao (for IMU), @AnthonyZhOon 
for the twizy: me, @AnthonyZhOon 

I'll try to make some docs in the wiki tonight

AbBaSaMo, 2023-12-06:
@dylan-gonzalez what did you mean by lidar doesn't power on? it seems to be on rn and the web portal working

dylan-gonzalez, 2023-12-06:
Oh fair dinkum

AbBaSaMo, 2023-12-06:
Running into issues launching mcav_autonomy/autonomy_launch. Issue pinned down to streedtrone.launch.xml lines 20 and 60. 

Tasks to debug include
- [ ] figuring out which of the two lines is causing the issue (launching kiss_icp or streedrone vehicle model)
- [ ] resolving lack of launch file/directory for kiss_icp esp if this is the issue
- [ ] resolving missing xacro dependency which is probably the issue

AbBaSaMo, 2023-12-06:
Getting the error 

[image]

@AnthonyZhOon or @dylan-gonzalez if I just want to run the various sensors should I be doing it by running the streetdrone.launc.xml or should I just launch them one by one and test their output?

Eventually the former will need to work anyway no? Seems the current issue is the fact that the roslaunch pkg does not exist for humble https://index.ros.org/p/roslaunch/#humble

dylan-gonzalez, 2023-12-06:
Is this on the twizy?

AbBaSaMo, 2023-12-06:
Yup on the Twizy though does that make a difference?

AbBaSaMo, 2023-12-06:
Running VLP16 driver on twizzy outputs the right message but there is no data. Lidar is humming so it seems to be on but not doing anything atm

AbBaSaMo, 2023-12-09:
Have been trying to get my head around how everything launches.

Based on the [launch docs](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/launch-autoware/sensing/), [this](https://github.com/Monash-Connected-Autonomous-Vehicle/mcav_autoware_sensor_kit_launch/tree/main/sample_sensor_kit_launch/launch) is where we define our launch files for both sensor drivers as well as sensing modules we want to use.

So for example, they have a launch file for pointcloud preprocessor. What remains then is implementing/modifying launch files for the following:
- [ ] gnss_poser
- [ ] imu_corrector
- [ ] vehicle_velocity_converter
- [ ] imu.launch.xml
- [ ] gnss.launch.xml

AbBaSaMo, 2023-12-09:
@dylan-gonzalez do we want to include the [gryo_bias_estimator](https://autowarefoundation.github.io/autoware.universe/main/sensing/imu_corrector/) package?

dylan-gonzalez, 2023-12-11:
If you guys think so?

dylan-gonzalez, 2023-12-11:
11-10-23

- make sure all sensors work on the van (individually)
- make sure all sensors work on the twizy (indivudally)
- once all vehicle sensors work, update @Robin-Chea so he can make changes to the mcav_sensor_kit_launch files
- look at radar sensor fusion with/wo lidar

zacharyeap, 2023-12-12:
RadarTracks
- messages convey info about tracks of detected objects through radio waves
- Contains more information about detected objects (eg. position, velocity, acceleration, size) 
- Objects represented by a unique ID generated by radar 
- All data of the object is relative to radar's frame of reference 

PointCloud 
- Represents set of datapoints in a 3d coordinate system 
- Each point usually includes coordinates (x,y,z) from the sensors frame of reference 
- The actual data is stored in a 1D array so the format and arrangement depends on specific fields (eg. height and width)