Monash-Connected-Autonomous-Vehicle/autoware #8: CI/CD pipeline

dylan-gonzalez, 2023-11-26:
- setup ci/cd pipeline (could use existing Autoware ci/cd pipeline)
   - take a look at this: https://github.com/Monash-Connected-Autonomous-Vehicle/mcav-autoware/tree/main/.github/workflows
- create test cases

sxppro, 2023-11-28:
i'll need to look into how they're running the ci/cd pipeline as well as how we build autoware

AnthonyZhOon, 2023-11-30:
We could start this off by meeting and sharing what we know about this, I'm on-campus weekdays but only have 1-2 hours until after 5pm

dylan-gonzalez, 2023-11-30:
Btw can you guys prioritise figuring out how the rest of the team can write test cases for their different components?

npnquang, 2023-12-01:
I found the folder storing the YAML files to run the CI/CD pipelines of Autoware. [These](https://github.com/autowarefoundation/autoware/tree/main/.github/workflows) are helpful to setup our own pipelines.

dylan-gonzalez, 2023-12-05:
Next steps:

- organise a working session @npnquang @AnthonyZhOon @sxppro 
- understand the different workflows for autoware, autoware.universe, autoware_launch, etc.
- test doing a push to a test branch --> try forcing an error/success

After that:
- see if we can launch AWSIM/CARLA from a workflow inside a docker container?

sxppro, 2023-12-07:
Okay right now we're looking at the `autoware.universe` pipeline — specifically the **build-and-test** workflow. We are currently running into an error:

```bash
jsk_recognition_msgs: Cannot locate rosdep definition for [rostest]
```

that is causing the rosdeps to not be successfully installed. 

Comparing our workflow run to Autoware's autoware.universe repo, they can install all the rosdeps successfully but I don't know why ours is different. We'll have to look more into this.

sxppro, 2023-12-07:
Some more info we've looked at:

- the `colcon build` action that runs in the pipeline: https://github.com/autowarefoundation/autoware-github-actions/tree/main/colcon-build
- the workflow also references a container image by Autoware, which also contains a nice overview of their repositories: https://github.com/autowarefoundation/autoware/pkgs/container/autoware-universe

sxppro, 2023-12-07:
We have identified the issue as `jsk_recognition_msgs` is a new package added by us that does not exist in the original repository. 

We likely just need to install rostest as a rosdep in the pipeline (which is not currently being done by the workflow definition).

npnquang, 2023-12-08:
I tried to find a way to install the `rostest` package that we are missing and then found [this](https://index.ros.org/p/rostest/github-ros-ros_comm/#humble). It says that there is no version of `rostest` for ROS Humble. I think we need to look into why we have `rostest` in the first place.

sxppro, 2023-12-11:
`rostest` is added as a _testdepend_ here: https://github.com/Monash-Connected-Autonomous-Vehicle/autoware.universe/blob/cbb62d06a84e40a91036effc458408e3b55364fe/sensing/jsk_recognition_msgs/package.xml

sxppro, 2023-12-11:
There's been some tests written with it: https://github.com/Monash-Connected-Autonomous-Vehicle/autoware.universe/tree/cbb62d06a84e40a91036effc458408e3b55364fe/sensing/jsk_recognition_msgs/test

dylan-gonzalez, 2023-12-11:
Can you see if anyone has made an issue about it on the autoware.universe repo?

Or alternatively check [this](https://autowarefoundation.github.io/autoware-documentation/main/contributing/testing-guidelines/unit-testing/) out and see if you can use g_tests instead??

dylan-gonzalez, 2023-12-11:
Hey @AnthonyZhOon I'm thinking that we some repository that stores all relevant dotfiles for the MCAV PCs (and for people doing mcav stuff on their own laptops) so that we can just automatically clone that repo onto the MCAV PCs and have all the settings like bashrc, vscode extensions, etc. configured to be synced? Can we set something up like that?

dylan-gonzalez, 2023-12-11:
also @AnthonyZhOon there's an autoware_lint_common package? Can someone look into that to see if we can use the linting that is already used in autoware?

sxppro, 2023-12-11:
@Robin-Chea can u remember where you added these tests from in this commit or did you write them yourself? 😮  https://github.com/Monash-Connected-Autonomous-Vehicle/autoware.universe/commit/92f7bfa0e12a1cd6bfa9c2fcdd9ae111e9664f40

Robin-Chea, 2023-12-11:
I dont remember :)

sxppro, 2023-12-11:
> I dont remember :)

surely u know whats the commit for right 😬

Robin-Chea, 2023-12-11:
😬

AnthonyZhOon, 2023-12-18:
Local install https://docs.astral.sh/ruff/tutorial/
Git page with docs https://github.com/astral-sh/ruff
Place a ruff.toml in the top-closest relevant directory, can do multiple in the file structure

dylan-gonzalez, 2023-12-18:
tomorrow night 19-12-23

@sxppro @AnthonyZhOon we'll go through some research on running headless simulations through the CI/CD pipeline