Monash-Connected-Autonomous-Vehicle/autoware #9: localization

dylan-gonzalez, 2023-11-26:
Understand [Autoware Localization](https://autowarefoundation.github.io/autoware.universe/main/localization/ekf_localizer/)

* configure launch file
* ensure test cases are working

dylan-gonzalez, 2023-11-29:
@jxcksonli @KehanRana do you guys wanna schedule a call to go over this? Or do you understand

jxcksonli, 2023-11-29:
hey dylan, we can come in on Friday to discuss this, i see you're in the workshop then, but will you be free within your timeslot 11am-3pm to discuss?

dylan-gonzalez, 2023-11-29:
Okay let's do 1pm, I'll make a gcal. Update the working days in notion if you'll be there in person

KehanRana, 2023-11-29:
Okay sounds good.

dylan-gonzalez, 2023-12-11:
11-12-23

To try some more testing with Autoware and AWSIM and see if the correct localisation nodes are being outputted

jxcksonli, 2023-12-12:
Update on 12/12/2023 (using e2elaunch)
- Gyro_Odometer: Matches the AVP Diagram 
- Stop Filter: Input is /localization/pose_twist_function_filter/kinematic_state, but AVP Diagram shows ekf_odom. Output is same as expected
- NDT Scan Matcher: Some topics do not exist when I run ros2 topic list, e.g sensing/lidar/filtered_points and localization/efk_pose_with_covariance, Input: sensing/gnss/pose_with_covariance matches
- EKF Localizer: Inputs match, Outputs are a bit different
- Pose Initialiser Node: Inputs match, Output matches

jxcksonli, 2023-12-19:
[image]
[image]
[image]
[image]
[image]

These are the screenshots from the RQT Graph

Also, we were quite confused because the config files have some of the param.yaml files but not ones such as ekf_localizer, pose_intializer (which can be found at https://github.com/autowarefoundation/autoware_launch/tree/main/autoware_launch/config/localization), but gyro_odometer and stop_filter aren't there. (Which is confusing to us because they also appeared on the RQT Graph but not in the launch/config files itself)
[image]
So should we add those other param.yaml files into our config launch because we cannot turn off the components if we can't find them?