Monash-Connected-Autonomous-Vehicle/autoware #17: CARLA setup

dylan-gonzalez, 2023-11-27:
CARLA Debian package only has support for Ubuntu 18.04 and Ubuntu 22.04

- [x] I want a setup for CARLA that works regardless of host OS. Follow [docker installation](https://carla.readthedocs.io/en/latest/build_docker/)
- [ ] Get Autoware Universe working with CARLA. See [these tutorials](https://www.youtube.com/watch?v=EFH-vVxn180) made by Hatem Darweesh. I believe he's made his own 'OpenPlanner' thing to get this to work, not too sure... find out what exactly it is.
- [ ] Get the Twizy working with CARLA for Hardware-in-the-loop (HIL) demo

Jokua, 2023-12-01:
when testing out the tutorial dylan posted, there will be a problem related to no octomap cmake or the like, so it can be solved by running ```sudo apt install ros-humble-octomap-ros```

[image]

dylan-gonzalez, 2023-12-01:
does the OpenPlanner tutorial use Ros humble?

Jokua, 2023-12-01:
ran into another problem when following the instructions in the carla-autoware demo

[image]

[image]

Jokua, 2023-12-01:
going to continue seeing if i can do that, but otherwise the zenoh carla thing dylan posted might be another option because i found the github page for it https://github.com/evshary/autoware_carla_launch

dylan-gonzalez, 2023-12-04:
so no errors so far??

Jokua, 2023-12-07:
ok, just now got the bridge built, no errors so far

[image]

Don't panic if the folder location doesn't look right, what i understood what rocker does is that it "loads" the folders from your host machine into the docker container, and even if it gets removed every change to those folders stays so you don't lose progress (i think so far)

Jokua, 2023-12-07:
building the autoware packages in another container now (fingers crossed it works bcs i didn't bother checking if it uses localhost or not but eh we'll cross that bridge when it gets to it)

[image]

Jokua, 2023-12-07:
both containers are now built, now only left to test the demo case to see if it works

Jokua, 2023-12-07:
ok, when running these commands for the demo, the map for carla did change, so at least that's working for now

[image]

Jokua, 2023-12-07:
but i'm not particularly sure when it's asking for the host1 ip (the pc running carla ) is the docker IP OR the actual host IP

Jokua, 2023-12-07:
yep, it works, and confirmed on both the rviz side and CARLA sim side, but it's really laggy so running in headless mode for CARLA should be more sensible.

dylan-gonzalez, 2023-12-11:
11-12-23

@Jokua to do some more testing with CARLA + Autoware-Universe

- try some parking scenarios
- update the [wiki](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware/wiki/Autoware) with CARLA launch params
- try multiple autoware vehicles?? (since it uses that zenoh bridge thing)

Jokua, 2023-12-15:
ok, a bit of an update on what I have done so far:

I have managed to reduce the burden on the pc by running CARLA in "headless mode", which means it's gonna run without the GUI so it saves us some more resources.

[image]

it can be done when creating a new docker container with this command ```docker run  -p 2000-2002:2000-2002  --cpuset-cpus="0-5"  --runtime=nvidia  --gpus 'all,"capabilities=graphics,utility,display,video,compute"'  -e DISPLAY=$DISPLAY  -v /tmp/.X11-unix:/tmp/.X11-unix  -it  carlasim/carla  ./CarlaUE4.sh -opengl $1 -RenderOffScreen -quality-level=Epic```

Also i've managed to make it run properly on my laptop (well once, will update if i run into any issues because there was some error that goes like ```sh: 1: xdg-user-dir: not found``` when trying to follow the tutorial for setting up carla in your local env in [this tutorial.](https://antc2lt.medium.com/carla-on-ubuntu-20-04-with-docker-5c2ccdfe2f71#:~:text=Setting%20up%20the%20local%20environment)

Jokua, 2023-12-15:
I have managed to run two instances as the demo has shown

[image]

I have also managed to get them both to move at the same time, though it is sort of laggy?

[image]

Also I don't know why it decided to stop running by itself after moving once, so that's probably another thing to make note of for troubleshooting later.

Additionally, there is a FAQ on the Zenoh's github page about how to change the map [here](https://github.com/evshary/autoware_carla_launch/blob/main/carla_map/README.md)

Still unsure on how to change the car's URDF or model, might have to look up on how carla spawns it or ask someone who has used CARLA before.

dylan-gonzalez, 2023-12-15:
> I have managed to reduce the burden on the pc by running CARLA in "headless mode", which means it's gonna run without the GUI so it saves us some more resources.

This would be very useful for the CI/CD pipeline guys @AnthonyZhOon @sxppro @npnquang 

> I have also managed to get them both to move at the same time, though it is sort of laggy?

If you record a rosbag and play it back again, is it still laggy?

Jokua, 2023-12-15:
I had to look up what a rosbag is, but how would I do that? 

Also I'll update the wiki on how to set it up and use it in the beast as well, since I've confirmed it can work on my pc (be warned it takes ages if your internet is shite to download the zenoh autoware bridge and autoware docker images)

dylan-gonzalez, 2023-12-27:
@Jokua if you're interested https://github.com/orgs/autowarefoundation/discussions/2828#discussioncomment-7929757

Jokua, 2023-12-27:
oh, thanks! I'll check it out, hopefully the guy updated some things

dylan-gonzalez, 2024-01-08:
@Jokua to do some more testing for the next 2 days

Jokua, 2024-01-10:
I have just finished some testing, seems like the problem I had about not being able to set another route may be just slight user error, apart from that recorded quite a lengthy video (with some cropping issues, i accidentally captures my other screen as well, as well as the fact i couldn't record my own voice)