Monash-Connected-Autonomous-Vehicle/autoware #32: CARLA x Autoware test on Twizy

dylan-gonzalez, 2024-01-07:


dtonda8, 2024-01-28:
[This partially resolved issue](https://github.com/hatem-darweesh/op_bridge/issues/27) is pretty much a tutorial for setting up Carla x Autoware (i.e. set a path in Rviz and Autoware Ackermann Control Messages are published to ROS, and thus, can control the Twizy).  

If you want to run the Carla x Autoware in one machine, and SD-VehicleInterface in another, you'll need to run the following commands on both machines' terminals:
```
export ROS_LOCALHOST_ONLY=2 # This can be any number
export ROS_DOMAIN_ID=2      # This can be any number, but must match the
```

dylan-gonzalez, 2024-01-29:
is there a repo for this?

dtonda8, 2024-01-29:
You'll need  3 [repos by hatem-darweesh](https://github.com/orgs/autowarefoundation/discussions/2828#discussioncomment-7929757). I managed to get it working on _the Beast_.

dylan-gonzalez, 2024-01-29:
It would be good if this could all be packaged in some way into one repository so that it's reproducible.
I've just made a new [repo](https://github.com/Monash-Connected-Autonomous-Vehicle/streetdrone-hil) for running HIL demos with the Streetdrone vehicles as I need to set up a CARLA HIL demo for a unit. 

Would be neat if (when you get time) make a branch for carla-autoware-zenoh bridge HIL demo and could probably add those 3 repos ^ as submodules?

dylan-gonzalez, 2024-01-29:
Also great work!!

closed 2024-01-29 by dylan-gonzalez

dtonda8, 2024-01-29:
Also Rviz + Carla should be done on a machine with at least 32 gb ram (i.e. not the SD Alienware Laptop). Maybe we could use one of the hives or something. Then we could publish to ROS topics via LAN.

dtonda8, 2024-02-12:
Rviz (on ~8 fps) + Carla running on the SD Alienware: https://drive.google.com/file/d/1ywcLC-7fMzx-sukkRjqpWB_ePQO9rQ5b/view?usp=share_link