Monash-Connected-Autonomous-Vehicle/autoware #33: Autoware installation on Twizy

dylan-gonzalez, 2024-01-07:


dylan-gonzalez, 2024-01-08:
@jxcksonli @KehanRana

dtonda8, 2024-01-22:
#### Alienware Anydesk
Remote Address: `1998290772`
Password: `cavisthefuture`

Robin-Chea, 2024-01-25:
Installing all dependencies using the anisible script 
./setup-dev-env.sh
[image]
missing python 3.10 venv so installed it with 

sudo apt install python3.10-venv

rerun ./setup-dev-env.sh

Robin-Chea, 2024-01-25:
[image]
going to install dependencies from https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/
[image]

dtonda8, 2024-04-17:
Pretty much follow the [source installation instructions](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/) and try to run some of the [planning sims](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/)

amrsnch, 2024-04-24:
Autoware is installed without any issues (probably bold to assume that it works, me and @tzdu will run some tests later this or next week to verify that). Original Autoware repo was used, and the solution provided should work for the MCAV fork as well (will check it later on a VM) 

1. Clone the repository

    **MCAV fork :** `git clone https://github.com/Monash-Connected-Autonomous-Vehicle/autoware.git`
    **Autoware original:** `https://github.com/autowarefoundation/autoware.git`

2. Run `./setup-dev-env.sh` while in `autoware` directory on Linux

3. It should give an error about the `ccache` failing, which is already there. So error can be ignored. 
    If `ccache` is not there, it should be installed manually. 

5. There are 2 ROS dependencies (one if you are using Autoware fork) that ruins the `rosdep install `, those 
    are:
              a. `autoware_internal_msgs` which can be fixed by copy-pasting the content of `autoware.repos` in 
                   Autoware fork into the `autoware.repos` in local directory (i.e on Linux) [I'll prly make a PR on 
                   that matter, so it is in our fork as well]
              b. `libqt5-charts-dev` can be downloaded via `sudo apt install libqt5charts5-dev`

7. After that `rodep install` should run appropriately without any warnings.

8. **The other important thing is to be careful with is the `colcon build`**. When you run `colcon build` on 
    Alienware it attempts to occupy all threads on the laptop, so it will lead to a deadlock in some point and 
    cause Alienware to crash. To resolve that add `--parallel-workers number-of-threads ` flag at the end. 
    Which basically sets a limit on treads that can be used by `colcon` (as far as I know the safe maximum is 
    8-10 threads). 
   Or just: `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 8`

amrsnch, 2024-04-29:
Error while truing to run autoware, reinstalling artifacts and rebuilding the environment

amrsnch, 2024-05-09:
Autoware is working now and sims run smoothly on Twizy. I had to resort to docker installation which is outlined here: [](https://autowarefoundation.github.io/autoware-documentation/galactic/installation/autoware/docker-installation/)

Pretty much follow the outlined instructions, however in some cases when you have several options, chose the following: 

- Clone main Autoware branch, not humble. Didn't try humble but seem to work fine
- Launch Docker container for amd64 architecture (might vary for different hadware specs)
- And again, don't forget to use `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release --parallel-workers 8`, not to crash the device.

Apart from that everything is straightforward

You will still need to download `libqt5-charts-dev via sudo `apt install libqt5charts5-dev`

And yeah, ignore ccache error while running ansible script.

Run `source ~/autoware/install/setup.bash` to verify that autoware has been installed correctly. And after that you can proceed to simulations planning

amrsnch, 2024-05-09:
Planning to run more sims in the following week with @tzdu

closed 2024-05-09 by amrsnch