Monash-Connected-Autonomous-Vehicle/autoware #37: Infrastructure mesh network

zcdai, 2024-02-05:
Setup a NetMaker server on cloud platform and dynamically connect/disconnect devices from it.

Farhan3135, 2024-02-05:
for domain registration should i use the email i used to get gcp or the auto gen one? 

Email Address for Domain Registration (click 'enter' to use ec757b34b810e256@email.com):

Farhan3135, 2024-02-08:
the command is:
netclient join -t <TOKEN>

I got netmaker working on the vm an netmaker on my pc

but how do i get the <TOKEN> to add my pc to the correct network?

@zcdai

zcdai, 2024-02-08:
> the command is: netclient join -t
> 
> I got netmaker working on the vm an netmaker on my pc
> 
> but how do i get the to add my pc to the correct network?
> 
> @zcdai

When you setup the vm to host the server, it should provide you with a URL that lets you access the GUI for the server.

In this GUI, you can create new node instances each with corresponding access tokens, these are the tokens you use to connect to the network.

If you are unsure of what URL lets you access the GUI, reinitiate NetMaker on the vm, and pay close attention to each line during instantiation.