Monash-Connected-Autonomous-Vehicle/autoware #38: (Twizy) Ensure Vehicle Interface publishes to Autoware

dtonda8, 2024-04-02:


dtonda8, 2024-07-09:
> [!IMPORTANT] 
> Twizy's battery dead. Do not work on this task until it's charged.

dtonda8, 2024-09-02:
Hey guys, this issue will require your hands dirty with the C++ Vehicle Interface (a tricky but rewarding experience). Fortunately, StreetDrone has given a guide of what sort of data you can read from CAN: [GDrive Link to User Manual](https://drive.google.com/file/d/1GlfjZdrJ_SD5M_kqQlOm5BciVqtmoW2Z/view).  p.18 onwards tells us what sort of data we can send & receive.  If we can receive CAN data from the Twizy that is requested by Autoware (e.g. `control_mode` & `steering_status`), then we can try testing/implementing it to SD-VehicleInterface repo.

Before coding, we'll need to ensure we can at check that we can at least read the CAN data, [useful notion docs on this](https://www.notion.so/monashcav/Reading-CAN-data-6ecdc23ba5ec4b38b121fd18c5187cad?pvs=4#51052367bd8a4aeda76078f26bf20da0). Once you found any useful CAN data according to the Twizy User Manual, lmk! (and will test CAN reading together)

dtonda8, 2024-10-01:
Hi all. Surya and I were integrating the publisher for `/vehicle/status/turn_indicators_status`. Adding a publisher is not easy - it involves ROS2, C++ and Autoware knowledge - so I wrote the guide below.

### Steps followed to add a new publisher:
1. Include relevant header file for the message, [e.g.](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L37). The header file is usually of the format `<package-name>/msg/<snake-case-msg-name>`. To confirm, search in the autoware.universe repo, [e.g.](https://github.com/search?q=repo%3Aautowarefoundation%2Fautoware.universe%20%22autoware_vehicle_msgs%2Fmsg%2Fturn_indicators_report.hpp%22&type=code).
2. Declare message, [e.g.](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L111)
3. Initialise publisher, [e.g.](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L124)
4. Update `main_loop` parameters to include message and publisher, [e.g.](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L130)
5. Perform your logic to populate message
6. Publish message [e.g.](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L213)
7. Ensure our vehicle_inteface package dependencies are added to package.xml ([e.g.1](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/package.xml#L18)) and CMakeLists.txt ([e.g.1](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/CMakeLists.txt#L19) & [e.g.2](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/CMakeLists.txt#L25))
  - Most messages (beside BatteryStatus as of 1/09/2024) fall under the autoware_vehicle_msgs, which is already added
  
### Testing:
1. There already exists a colcon workspace with all dependencies. Navigate to `/home/mcav/test_sd/sept_auto_publisher_ws/src/SD-VehicleInterface` on the Alienware Laptop
  - Otherwise you can create your own ros workspace with [SD-VehicleInterface](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/tree/cdccad164580d2ee09fd8866bdc9b82d351c6516) and [autoware_msgs](https://github.com/autowarefoundation/autoware_msgs) cloned into `src/`
2. In the SD-VehicleInterface which is located on `src`, `git pull` and ensure you're on the correct branch
3. `colcon build`, debugging errors as necessary
4. [Launch Vehicle Interface](https://github.com/Monash-Connected-Autonomous-Vehicle/twizy-hil/tree/main#twizy-hil), providing the correct workspace path to the `first.sh` script
  - Optionally run simulation mode if you're planning to test dummy publishing: `ros2 launch sd_vehicle_interface sd_vehicle_interface.launch.xml sd_vehicle:=twizy sd_speed_source:=vehicle_can_speed sd_simulation_mode:=true`
5. Check if the Vehicle Interface actual publishes to the topic with `ros2 topic echo <autoware-topic>`

### What's left for `/vehicle/status/turn_indicators_status`:
- @Surya-Sathya , You might need to analyse frame 0x117 ([deets on p.g. 39](https://drive.google.com/drive/u/1/search?q=twizy%20user%20manual)) instead of frame 0x104. Frame 0x117 is a report from the twizy, whereas 0x104 is where we control the twizy.
- Then, we can use Surya's CAN parser to correctly populate the [indicator status message here](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/blob/cdccad164580d2ee09fd8866bdc9b82d351c6516/vehicle_interface/src/sd_vehicle_interface/sd_vehicle_interface.cpp#L212C3-L212C27)
  - [`TurnIndicatorsReport`Message](https://github.com/autowarefoundation/autoware_msgs/blob/main/autoware_vehicle_msgs/msg/TurnIndicatorsReport.msg) details what constants you can use and what they represent
  
### Note:
1. Surya and I noticed that the Autoware documentation link to wrong repositories. The [autoware_msgs](https://github.com/autowarefoundation/autoware_msgs) is the correct repo of all messages.
2. I noticed that I cannot parse StreetDrone_Data_2 (the frame conaintaing info about indicators) nor Customer_Control_2 (the frame to control the car).

If in doubt, add a comment to this issue or dm me on Slack.

Surya-Sathya, 2025-04-10:
**Testing Detailed Instructions**
Purpose: We want to ensure that the vehicle interface is publishing its status via ROS to Autoware

Pre-testing: Turn on Twizzy via auxiliary (Wifi network name: Streetdone) **Do not connect to eduroam**. Use Alienware laptop or Dell laptop and connect to this network. 

1. Create a directory (mkdir test_publish_workspace after making an april month pacakge)
2. Clone into this directory (after creating src directory) (https://github.com/streetdrone-home/SD-VehicleInterface) and clone both (https://github.com/autowarefoundation/autoware_msgs) into the source directory. For now use Surya 
3. Now we should have april/testpublish/src -> Doing ls -> _ackermanSuryaTesting autoware msgs_
5. run: <colcon build> (C++ files into machine code -> if there's any bugs with typos and errors in the code, then it colcon should fail to build) If the build fails, find errors in vehicle interface and 

Currently: Not building because unused frame variables. Understand in vehicle itnerface what is happending with publishers, and use frame data from indicators and populate main loop eventually