Monash-Connected-Autonomous-Vehicle/autoware #40: (Twizy) Ensure sensors work

dtonda8, 2024-04-02:
- x2 people
**Goal**: Collate/create/find launch files (`.xml` or `.launch.py`) for launching sensors

Ensure sensors work and publish correct Message Types:
- [ ] LiDAR as `sensor_msgs/PointCloud2.msg` and launch file 
- [ ] IMU as `sensor_msgs/Imu.msg`
- [ ] GNSS as `sensor_msgs/NavSatFix.msg` and `autoware_sensing_msgs/GnssInsOrientationStamped`
- [ ] Radar as `sensor_msgs/PointCloud2.msg`
- [ ] Cameras as `sensor_msgs/Image.msg`

Helpful resources
- [Autoware's *Sensor launch*](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/creating-sensor-model/#sensor-launch)
- [*Sensor component design*](https://autowarefoundation.github.io/autoware-documentation/main/design/autoware-architecture/sensing/#high-level-architecture) details more about each sensor
- Consult this [unresolved issue](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware/issues/7) for more details about what's currently working

dtonda8, 2024-04-19:
SSH
```
ssh mcav@192.168.11.127
```

password: `mcav01`

dtonda8, 2024-04-19:
For Velodyne, Alienware Laptop and LIDAR must be connected to in-built router

dtonda8, 2024-04-19:
[Docs for starting the Vehicle Interface](https://github.com/Monash-Connected-Autonomous-Vehicle/twist_controller/blob/main/README.md#using-sd-vehicleinterface-with-twist-controller-the-following-is-mainly-for-the-twizy)

dtonda8, 2024-04-26:
https://github.com/Monash-Connected-Autonomous-Vehicle/stream_sample_camera_gmsl/tree/main

dtonda8, 2024-05-02:
[gscam2](https://github.com/clydemcqueen/gscam2) might useful

jxcksonli, 2024-05-02:
Attempt using gscam2, located in
mcav@twizy-x86:~/dbat/test1$

Ran using
ros2 run gscam2 gscam_main --ros-args --remap /image_raw:=/my_camera/image_raw --params-file src/gscam2/gscam_params.yaml -p camera_info_url:=file://$PWD/src/gscam2/cfg/my_camera.ini

Error: Failed to Pause Stream, Check GScam config

dtonda8, 2024-05-03:
[ROS answer on h264 streaming](https://answers.ros.org/question/379644/compress-image-topic-with-h264-streaming-for-low-bandwidth-connection/) i.e. what we want

dtonda8, 2024-05-05:
Hi guys gscam2 works! I've tried in my laptop. The tasks for you guys is to:

- [ ] Create a `camera.launch.xml` or `.py` launch file that runs the script. If you remember from last Thursday, gscam required a local workspace, so modify the script to source the correct workspace location.
- [ ] Create a `lidar.launch.xml` or `.py` launch file to launch the velodyne that we tested a 2 weeks ago

We won't need to create launch files for IMU since that's already done by the vehicle interface

#### `gscam` details
- [Installation](https://github.com/clydemcqueen/gscam2?tab=readme-ov-file#install-and-build). Then
```sh
source install/local_setup.bash && export GSCAM_CONFIG='udpsrc port=5000 caps="application/x-rtp" ! rtpjitterbuffer ! rtph264depay ! avdec_h264 ! videoconvert' && ros2 run gscam2 gscam_main --ros-args -p sync_sink:=true -p preroll:=true
```

[ROS 2 tutorial on creating launch files](https://docs.ros.org/en/humble/Tutorials/Intermediate/Launch/Creating-Launch-Files.html)

dtonda8, 2024-07-08:
Fortunately, the [default lidar.launch.xml](https://github.com/autowarefoundation/sample_sensor_kit_launch/blob/main/sample_sensor_kit_launch/launch/lidar.launch.xml) pretty much managed the launching. Just needed to tweak the default parameters, our resulting sensor kit launch is [here](https://github.com/dtonda8/twizy_sensor_kit_launch/tree/main).

To try out (as of our Autoware's current directory):
```bash
cd ~/twizy/autoware.twizy
source install/setup.bash
ros2 launch twizy_sensor_kit_launch sensing.launch.xml vehicle_mirror_param_file:=/home/mcav/twizy/autoware.twizy/src/vehicle/twizy_vehicle_launch/twizy_vehicle_description/config/mirror.param.yaml vehicle_param_file:=/home/mcav/twizy/autoware.twizy/src/vehicle/twizy_vehicle_launch/twizy_vehicle_description/config/vehicle_info.param.yaml
```

 The working tweaks of the Lidar settings at http://192.168.1.201/ and alienware Wired interface are shown below:

closed 2024-07-09 by dtonda8