Monash-Connected-Autonomous-Vehicle/autoware #46: Simplify running simulator scenarios

dtonda8, 2024-07-09:
Working with Autoware planning simulations can be a little figety. We want to simplify this, especially when performing demos.
We could either:
1. Use an extensive [Simulator Scenario](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/scenario-test-simulation/) ROS package to do this. And combine commands to shell scrips
2. Or, could provide a command line that takes that provides a list of scenarios (e.g. Road to Parking, Car to Obstacle, Reversing) we could automate the demo process. E.g.
```
What scenario would you like to demo?
1. Parking
2. Car stopping due to random obstacle
3. A to B then back to B
4. etc.

Input: 
```

For the second option:
- You'll need to first know the starting/ending locations for each scenario
- For each scenario, the new ROS package first sets the vehicle to the starting point and sets the end point
- To get an idea of topics published, I recommend following the default [Autoware Planning Simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/), then running `ros2 topic list` & `ros2 topic echo`

Obviously the first option sounds simipler so I encourage you to try that first, but if it over complicates our workspace than you can opt to option 2.

Skills: Python/C++ & ROS or Shell scripting.

dtonda8, 2024-09-02:
Btw the first thing I linked above assumes [Autoware is installed](https://autowarefoundation.github.io/autoware-documentation/main/installation/autoware/source-installation/) as well as [`simulator.repos`](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/scenario-simulation/planning-simulation/installation/)

dtonda8, 2024-09-08:
For option 1, I've posted an new issue on the scenario_simlutor_v2 here: https://github.com/tier4/scenario_simulator_v2/issues/1365. We will wait for a maintainer to respond, but if any of you guys managed to get the the [scenario_test_runner](https://tier4.github.io/scenario_simulator_v2-docs/user_guide/QuickStart/#scenario_test_runner), let me know.

dtonda8, 2024-09-22:
Got some good advice about from the scenario_simulator_v2 issue above. I managed to get it working on my 8 GB RAM Ubuntu VM. So, it should work decently on the Alienware.

Another idea for storing scenarios is to use [`rosbag`](http://wiki.ros.org/rosbag/Commandline). `rosbag` essentially allow you to record the ROS messages and play them back. So, we could potentially record our manually planning simulations, save them to rosbags, and play them at any time we want without getting fidgety with the Autoware/RViz GUI.

We might get back to this task once you guys are done with the cameras.

emilylkha, 2024-09-27:
Repository with info on how to run it: https://github.com/Monash-Connected-Autonomous-Vehicle/simulator-scenario-scripts