Monash-Connected-Autonomous-Vehicle/autoware #47: (ENV200) Automate start up

dtonda8, 2024-07-09:
I noticed it's quite tedioous to start the computer of the Van (networking, ssh prep etc.)
I think it's worth investigating various ways to optimise this, we could either try to properly install ubuntu or look at start up skills.

Skills: Shell scripting, Linux OS & a bit of Networking

dtonda8, 2024-07-10:
Hey Luke & Quang. I think this task is worth looking at first before Autoware installation:
Options:
1. Try to use the Desktop Image of [Ubuntu 22](https://releases.ubuntu.com/jammy/) → We then won't need to use xRDP
2. Create a `sh` script that sets up up networking, ssh etc. That is, a random UA software member should be able to use Ubuntu normally (with internet access) without needing to know the settings 

Lmk if you need assistance. Post updates/logs, helpful resources and anything else useful on this GitHub Issue.

LJ2533, 2024-07-18:
I haven't been able to come in for the past week but I'll be in tomorrow to work on this task.
Having the desktop version of Ubuntu should improve ease of use so I plan on implementing that. I'll update tomorrow once this is complete.

dtonda8, 2024-07-18:
Thanks for the update! I'll be on campus till 2pm tmr, let me know if you're facing any issues.

LJ2533, 2024-07-19:
Jin and I have installed the desktop image for the van and its working nicely. We did a quick test of the NVidia drivers as well, as that was went wrong last time and there were no problems there either.

dtonda8, 2024-07-19:
Nice work guys! How's the networking though? Is it easy to connect to wifi after `sudo reboot`?

closed 2024-07-19 by dtonda8