Monash-Connected-Autonomous-Vehicle/autoware #48: (Twizy) Intrinsic Camera Calibration

dtonda8, 2024-07-09:
[Relevant Autoware Docs](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/calibrating-sensors/intrinsic-camera-calibration/)

dtonda8, 2024-07-11:
Hi all! You don't need to worry about starting the Camera since is already on (as of 11th of July). Camera feed can be viewed by running the following on the Alienware ([you can anydesk to it](https://www.notion.so/monashcav/Twizy-3dc0bf8de24c4e38acb5b78ab59edda5)):

```bash
cd ~/autoware.twizy
source install/setup.bash
ros2 launch twizy_sensor_kit_launch camera.launch.py
```

If it doesn't work, someone on campus needs to start it up again (lmk if that's the case)

dtonda8, 2024-07-19:
Hey Thomas, how's your progress so far?

The Camera is back up running. Use the command above to launch the cameras. You can view the camera feed using `rviz2` in terminal.

tzdu, 2024-07-24:
Hey David, I'm so sorry about the delays, I will be calibrating the camera today.

tzdu, 2024-07-24:
[image]

Hey David, when I try running the command I get the following issue

dtonda8, 2024-07-24:
@tzdu that makes sense since cameras are down now (I might turn the cameras back on on Friday). Try attending our next UA meeting and we can try running the commands together.

I'll investigate how to make it easier to turn on the cameras. Thanks for trying the commands tho

tzdu, 2024-07-24:
Ok cool thanks. Also with the calibration chessboard, will printing one for use be fine?

dtonda8, 2024-07-24:
Yea that's a good idea, the docs might have the correct sizing required

dtonda8, 2024-08-05:
Starting Camera feed to be viewable from Alienware Laptop
- Ensure the PX 2 and Alienware Laptop is connected to the StreetDrone Router (via Ethernet). The Alienware should have the "Wired Connection" on the top right
- Ensure you can ping the same I.P. address as the one found in `~/Downloads/sender_alienware.sh`. If not working, update the script with correct IP address of the Alienware Laptop

Instructions
1. Turn on PX 2
2. Login to nvidia (password: `nvidia`)
3. Open terminal and run the following
```bash
cd ~/Downloads
./sender_alienware.sh
```
You should now be able to view the camera feed via the PX 2 Display

5. On the Alienware
```bash
cd ~/autoware.twizy
source install/setup.bash
ros2 launch twizy_sensor_kit_launch camera.launch.py
```

6. To verify if camera feed is actually streaming, open rviz from terminal: `rviz2`
7. Add the camera topic 
8. If you cannot see camera topic, then its msot likely the case that streaming is not working

Feel free to update the instructions above so it can be more clear.

tzdu, 2024-08-05:
@dtonda8 Hey David, when I run the sender_alienware file it shows one of the cameras but it's a blocked one. Am I using the wrong camera? And if so how do I switch it?

dtonda8, 2024-08-05:
Chen was testing cameras, and had to tape some of them 😆 . Cameras are working tho, prolly best to test cameras while campus.

tzdu, 2024-08-08:
Calibrated the camera and saved the file as intrinsic_camera_calibration_info.yaml in twizy/twizy_sensor_kit_launch and twizy/twizy_individual_params. 

Also pushed the files to dtonda8/twizy_sensor_kit_launch  and dtonda8/twizy_individual_params (cloned to dbat folder on alienware).  

Need to check which twizy_sensor_kit_launch and twizy_individual_params to push the files to.

dtonda8, 2024-08-12:
Individual params seems to be the correct one, we'll fix it later if its not. Feel free to close issue once you're done!

closed 2024-08-12 by tzdu