Monash-Connected-Autonomous-Vehicle/autoware #49: (Twizy) Start PX 2 Camera streaming remotely

dtonda8, 2024-07-09:
Currently, the camera stream needs to be started locally (i.e. log in to PX 2 and start camera stream). 

We want to be about start the camera stream automatically. 
- [ ] Either use `ssh` via alienware laptop (previously tried)
- [ ] Or, start camera stream automatically on PX 2 boot up

While this is not necessary, this automation would save a lot of time

dtonda8, 2024-07-19:
Hey Daniel, how's your progress so far?

This might be a good start of running the `sh` script without needing to login: https://askubuntu.com/questions/745933/execute-script-before-login

hong0431, 2024-07-19:
Hi David,
Not too much, I'll have a lot more time to work on it and figure it out before tmr's meeting.

hong0431, 2024-07-23:
We've tried multiple methods to get the camera automatically startup with no luck.

Link for different methods used for future reference:
https://www.squash.io/executing-bash-script-at-startup-in-ubuntu-linux/

closed 2024-07-23 by hong0431

dtonda8, 2024-07-23:
Thanks @hong0431. I've created another task that's sort of related: https://github.com/orgs/Monash-Connected-Autonomous-Vehicle/projects/8/views/15?pane=issue&itemId=71519558

It's more research based, have a look at notion docs, github and link any useful info.

dylan-gonzalez, 2024-07-23:
this shouldn't be too hard if you make a `systemd` service for it. https://askubuntu.com/questions/919054/how-do-i-run-a-single-command-at-startup-using-systemd

dtonda8, 2024-07-23:
I think @hong0431 tried the `systemd` service. We'll give it another shot tho.