Monash-Connected-Autonomous-Vehicle/autoware #50: (Twizy) Initial sensor calibration using CAD

dtonda8, 2024-07-09:
[Relevant Autoware Docs](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/calibrating-sensors/calibration-tools/#setting-of-sensor_kit_base_link-position-with-respect-to-the-base_link)

- [ ] Search for any existing CAD files of the Twizy in our shared Google Drive, Notion or in [streetdrone-home's github](https://github.com/streetdrone-home)
    - If no CAD files can be found, research how to convert urdf to CAD files (`.stl` files ?)
- [ ] Obtain measurements and update files as noted above
- [ ] Push calibrated `yaml` files to [twizy_sensor_kit_launch](https://github.com/dtonda8/twizy_sensor_kit_launch/blob/main/twizy_sensor_kit_description/config/sensors_calibration.yaml) and [twizy_individual_params](https://github.com/dtonda8/twizy_individual_params/blob/main/individual_params/config/twizy/twizy_sensor_kit/sensors_calibration.yaml)

dtonda8, 2024-07-10:
Hey Zhenlin.
I found a some [Twizy STL files](https://github.com/Monash-Connected-Autonomous-Vehicle/mcav_autonomy/tree/fefa5813668963790d07fd2720433d7ee8b7ef46/visualisation/vehicle_model/meshes) on one of MCAV's repos.

You familiar with using CAD to get measurements?

Lmk if you need assistance. Post updates/logs, helpful resources and anything else useful on this GitHub Issue.

dtonda8, 2024-07-19:
@zheeen Did you manage to try some of the STL files above? Try using CAD to get required measurements shown [here](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-and-sensor-model/calibrating-sensors/calibration-tools/#setting-of-sensor_kit_base_link-position-with-respect-to-the-base_link).

zheeen, 2024-07-19:
Hi David,
I've opened the STL files in SolidWorks, but I think the STL model of Twizy is scaled, currently just unsure of how they scaled the model. 
I'm also a little unsure about where the sensor kit is located on Twizy (is it just the LiDar sensor).

zheeen, 2024-07-22:
ratio for model to real life is 1:10