Monash-Connected-Autonomous-Vehicle/autoware #51: (ENV200) Fine-tuning Vehicle Interface

dtonda8, 2024-07-10:


jin-ramen, 2024-07-19:
[image]

dtonda8, 2024-07-19:
From these [docs](https://www.notion.so/monashcav/Reading-CAN-data-6ecdc23ba5ec4b38b121fd18c5187cad#c0c756b78f70487b822c00b1ccb008a4):

Try running this:
```bash
sudo modprobe peak_usb
sudo ip link set can0 up type can bitrate 500000
```
To set up can0

Then try reading the can with:
```bash
candump can0
```

If you're getting a large output → can is working. But if it's not outputing anything then there's another issue. Let's us know what you get.

jin-ramen, 2024-07-19:
Issues on entering autonomous mode
[image]

dtonda8, 2024-07-19:
This might be useful (from page 6):

jin-ramen, 2024-07-22:
@dtonda8 the issue is even when the turn the mode selection switch the light still stays green (manual). The only thing that changes is the "GCU_A_State" which turns from "MANUAL" -> "AUTO" when we turn the mode selection switch to right

dtonda8, 2024-07-22:
@jin-ramen 
Weird... you can try the following trouble shooting steps from ENV200 docs: 

[Link if you cannot see the image](https://imgur.com/a/ivtiwbK)
[image]

Otherwise you can have a look if you written any docs previously. I'll let you know if I find any useful info

jin-ramen, 2024-07-22:
One possible reason is that the van needs to be "park[ed] on a flat even surface" which the van isn't.

dtonda8, 2024-07-22:
Hm... You've managed to get the van running with twist msgs before, right?

jin-ramen, 2024-07-22:
Nah never done it before

jin-ramen, 2024-07-22:
Lights now goes yellow when mode selection switch is turned to the right. 

However, it does not stays yellow after a handshake is made.

Potentially CAN problem.
Following are outputted when vehicle_interface is launched:
```
mcav@mcav:~/test_sd_ws$ ./launch_real.sh 
CAN bus setup successful
[INFO] [launch]: All log files can be found below /home/mcav/.ros/log/2024-07-22-04-06-29-472202-mcav-2300
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [sd_vehicle_interface_node-1]: process started with pid [2311]
[INFO] [socket_can_receiver_node_exe-2]: process started with pid [2313]
[INFO] [socket_can_sender_node_exe-3]: process started with pid [2315]
[socket_can_receiver_node_exe-2] [INFO] [1721621190.086192055] [socket_can_receiver]: interface: can0
[socket_can_receiver_node_exe-2] [INFO] [1721621190.086275997] [socket_can_receiver]: use bus time: 0
[socket_can_receiver_node_exe-2] [INFO] [1721621190.086289320] [socket_can_receiver]: can fd enabled: false
[socket_can_receiver_node_exe-2] [INFO] [1721621190.086298645] [socket_can_receiver]: interval(s): 0.010000
[socket_can_sender_node_exe-3] [INFO] [1721621190.096647332] [socket_can_sender]: interface: can0
[socket_can_sender_node_exe-3] [INFO] [1721621190.096729000] [socket_can_sender]: can fd enabled: false
[socket_can_sender_node_exe-3] [INFO] [1721621190.096751602] [socket_can_sender]: timeout(s): 0.010000
[socket_can_receiver_node_exe-2] [INFO] [1721621190.296299551] [socket_can_receiver]: applied filters: 0:0
[socket_can_receiver_node_exe-2] [WARN] [1721621435.841018518] [rcl_lifecycle]: No transition matching 1 found for current state active
[socket_can_receiver_node_exe-2] [ERROR] [1721621435.841057562] [socket_can_receiver]: Unable to start transition 1 from current state active: Transition is not registered., at ./src/rcl_lifecycle.c:355
[socket_can_sender_node_exe-3] [WARN] [1721621435.842876314] [rcl_lifecycle]: No transition matching 1 found for current state active
[socket_can_sender_node_exe-3] [ERROR] [1721621435.842899567] [socket_can_sender]: Unable to start transition 1 from current state active: Transition is not registered., at ./src/rcl_lifecycle.c:355
```

jin-ramen, 2024-07-27:
Every time vehicle_interface is launched (`./launch_real.sh`), leads to CAN error, leading to not being able to enter autonomous mode. 

[image]

dtonda8, 2024-07-27:
I don't think we tried this yet lol. This should work:
[image]

We could also try  setting up can2 interface instead of can0 (i.e. modifying [instances of `can0`](https://github.com/search?q=repo%3AMonash-Connected-Autonomous-Vehicle%2FSD-VehicleInterface%20can0&type=code) to be `can2`)

If all of the above does not work, we'll contact SD

jin-ramen, 2024-07-29:
Activating Autonomous Mode: 
https://imgur.com/a/IP6LLni

jin-ramen, 2024-08-05:
We are having trouble turning into autonomous mode. Following the steps from the manual after turning the black mode switch, we launched CAN and vehicle_interface and published a twist msg but it always goes back to green light. 

We also tried turning on the controller switches, which results in flashing green and yellow light, and when we put into Drive mode, only the right front wheels move but the left front wheel does not move unless we pull the trigger in the controller. 

A request has been sent to [streetdrone](https://streetdrone.zendesk.com/hc/en-us)

jin-ramen, 2024-08-14:
The message can be found [here](https://streetdrone.zendesk.com/hc/en-us/requests/280)

dtonda8, 2024-08-14:
Thanks for doing that @jin-ramen! Can you still control the speed using the handheld controller? If that works that the twist msgs should work as well 🤔

jin-ramen, 2024-08-14:
@dtonda8 Controlling the vehicle with the controller works. It is possible that the problem lies in CAN bus where we are unable to publish twist or could be something wrong with the code. I'll look into that.

dtonda8, 2024-08-14:
Yep, so it must be the code then. For debugging you can try std::cout and print out can msgs to see if the twist msgs actually influence the can.

dtonda8, 2024-10-29:
Good news! Activating Autonomous mode via CAN is working now.

1. Ensure the computer is connected via CAN Adapter as shown no the image below. The hand held controller does not need to be connected

[image]

[image]

2. Follow [these instructions](https://www.notion.so/monashcav/Activating-Autonomous-Mode-f139c1d00b154074a9288365b43e26c2) **up to and including step 4**
3. Run the following on the connected computer
```bash
cd ~/test_sd/ack_ws

sudo modprobe peak_usb
sudo ip link set down can0 && sudo ip link set can0 up type can bitrate 500000

export ROS_LOCALHOST_ONLY=2
export ROS_DOMAIN_ID=2

source install/setup.bash

ros2 launch sd_vehicle_interface sd_vehicle_interface.launch.xml sd_vehicle:=env200 sd_speed_source:=vehicle_can_speed sd_simulation_mode:=false
```

The `ack_ws` has the following packages:
- SD-VehicleInterface, [ackermann](https://github.com/Monash-Connected-Autonomous-Vehicle/SD-VehicleInterface/tree/ackermann) branch
- [autoware-msgs](https://github.com/autowarefoundation/autoware_msgs.git)
- (Optional) twist_controller, [ackerman_controller](https://github.com/Monash-Connected-Autonomous-Vehicle/twist_controller/tree/ackermann_controller) branch. This is optional if you want to send ros2 topics.

5. Follow Step 5 then Step 7 of the same notion doc [instructions](https://www.notion.so/monashcav/Activating-Autonomous-Mode-f139c1d00b154074a9288365b43e26c2) (skipping Step 6 since we're not using the hand held controller)
6. (Optional) To test sending ros2 topics, on another terminal run:

```bash
cd ~/test_sd/ack_ws

export ROS_LOCALHOST_ONLY=2
export ROS_DOMAIN_ID=2

source install/setup.bash

ros2 run twist_controller twist_controller
```