Monash-Connected-Autonomous-Vehicle/autoware #58: (Twizy) Research Hardware replacement

dtonda8, 2024-11-04:


dtonda8, 2024-11-05:
Hi all! Add your name to the tasks you want to do (edit the post above then add (`<your-name>`) beside any task `- [ ]`). This should be chill, we should aim to finish this research within the next 7 days and add any useful things to this GitHub issue.

amrsnch, 2024-11-06:
Yo, I was wondering if we can go with a refurbished machine to save some money. Also, considering that it's going to be stored in the back of the Twizy and not used directly, it kind of makes sense. However, it also depends on the budget we have to spend, lol. If we can afford a new machine, we can go with that as well. A PC is also under consideration.

amrsnch, 2024-11-06:
Also, I've got a recommendation regarding Raspberry alternatives. You can check them out `here (https://www.khadas.com/vim4)`

dtonda8, 2024-11-06:
> Yo, I was wondering if we can go with a refurbished machine to save some money. Also, considering that it's going to be stored in the back of the Twizy and not used directly, it kind of makes sense.

Yep, definitely. It could even have a broken screen since it would still be usable via Anydesk or ssh

> However, it also depends on the budget we have to spend, lol. If we can afford a new machine, we can go with that as well. A PC is also under consideration.

Pros of PC: cheap and easier to upgrade. Cons: constant battery consumption and size.

> Also, I've got a recommendation regarding Raspberry alternatives. You can check them out here: https://www.khadas.com/vim4

This seems pretty interesting. We could have one these to solely manage the lidar or cameras. I'll think about this for a bit. Good find!

dtonda8, 2024-11-08:
The fyp teams have several Raspberry Pis
[image]
[image]

And there is a 16GB RAM laptop with Ubuntu, now in G03
[image]

dtonda8, 2024-11-18:
Fixed issue on why we couldn't ping the PX 2 from the new Dell Latitude laptop:
- [IP address wasn't in the same subnet](https://superuser.com/a/375165) as the px2. (Latitude: **192.168.10**.100, PX 2: **192.168.1**.181)
- Fixed with `sudo ip addr add 192.168.1.100/24 dev enp0s31f6`, i.e. changing ip address of interface to have same subnet as px 2 ([source](https://askubuntu.com/a/1243748))

I'll leave it to one of you guys to try running sensors from this laptop. I'll hold off from buying a new computer.