Monash-Connected-Autonomous-Vehicle/autoware #59: Deploy & tweak open-source Recruitment Platform

dtonda8, 2024-11-24:
One of the FIT3170 projects includes a student team recruitment platform: Inspo: https://github.com/Monash-FIT3170/Onboarder

We'll need to complete the following tasks in order as below:
- [ ] Check if licensing allows us to directly use it
- [ ] Fork repo into the MCAV github org
- [ ] Investigate the costs to run this site
- [ ] Let UA leads know once you have the estimate of running the site
- [ ] If the given the go, begin deployment
- [ ] Improve the site UI/UX

dtonda8, 2024-11-24:
Cherian spotted this, first sub-task done:
[image]

dtonda8, 2024-11-24:
Our own repo fork:
https://github.com/Monash-Connected-Autonomous-Vehicle/Onboarder

dtonda8, 2024-11-24:
What's done so far:
Supabase: Done
Encryption keys: Done (To be added to [env](https://drive.google.com/drive/u/1/folders/1wIfxJ_yS7WeUooKft93_Ck2tgFDHn4wp) soon)

TODO:
- [ ] AWS setup
- [ ] Google calendar URIs

amrsnch, 2024-12-01:
I can take of AWS set up

dtonda8, 2024-12-08:
Trello board for recruitment platform: https://trello.com/invite/b/67552eabb7ecc661bdc99080/ATTI3108e77324f1cbc306e28ea8d0725c180A041659/onboarder

dtonda8, 2024-12-15:
Managed to test deployment on http://13.236.193.216. Only "Apply for a position" button is functional since that's the only backend I've implemented with Flask

Deployed using aws ec2. One instance each for the frontend and backend.

Latest envs have been uploaded [here](https://drive.google.com/drive/u/1/folders/1wIfxJ_yS7WeUooKft93_Ck2tgFDHn4wp)

amrsnch, 2024-12-23:
Yo. I've added the MCAV logo to the Navbar and Login Screen. To enable inserting .svg images as React components, I integrated a new [vite plug-in](https://github.com/pd4d10/vite-plugin-svgr#usage). Additionally, I updated the theme colour to match our blue logo (hex: #0077ff).

Light mode: 
[image]

Dark mode:
[image]

Lmk if you have any suggestions regarding design aspects.

dtonda8, 2024-12-25:
Awesome @amrsnch! I like how MCAV logo matches with the light/dark mode. I've included you changes to our deployment.

For everyone:
I've fixed the OAuth login problems on deployment: http://13.236.193.216/ . You can locally deploy as well:  update to use latest [env](https://drive.google.com/drive/u/1/folders/1wIfxJ_yS7WeUooKft93_Ck2tgFDHn4wp) and run flask according to readme.md.

After login, most of the buttons aren't functional, we're still in the process of converting the original backend (AWS lambda) to use flask. If anyone else is interested then checkout the `flask_server/app.py`.