Monash-Connected-Autonomous-Vehicle/autoware #66: (ENV200) Integrate Ancillaries control with Autoware

dtonda8, 2025-04-01:
Ancillaries: Secondary components such as the horn, indicators, beam etc.
Earlier this year we managed to send CAN frames to control these components. We now want to connect these parts with Autoware.

- SD User manuals: https://drive.google.com/drive/u/0/folders/1jsNIZflh_5vnb5bbbD2K-ZQ8mYmCXrQ5

We'll need to make the vehicle interface (in `ackermann` branch) subscribes to necessary topics and sends CAN msgs to actually control the Vehicle:
- [ ] `/control/command/control_cmd` (already subscribing but not using steering_tire_rotation_rate)
- [ ] `/control/command/gear_cmd`
- [ ] `/control/current_gate_mode` 
- [ ] `/control/command/emergency_cmd`
- [ ] `/control/command/turn_indicators_cmd`
- [ ] `/control/command/hazard_lights_cmd`
- [ ] `/control/command/actuation_cmd`

[Source](https://autowarefoundation.github.io/autoware-documentation/main/how-to-guides/integrating-autoware/creating-vehicle-interface-package/creating-vehicle-interface/#creating-vehicle-interface) (for topics and descriptions). 

We've got some working examples of Autoware topic subscription on the `sd_vehicle_interface.cpp` file. E.g.:

[image]

You can get creative in how we test this sort of stuff. What've done in the past is that we would create another ROS2 package just to simulate these messages. Another thing we can do is indirectly control these via the (like making it turn right on the left on the simulation to trigger the left indicator), link to [planning sim](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/planning-simulation/#3-set-a-goal-pose-for-the-ego-vehicle)

shen-kit, 2025-04-17:
Added subscribers for `/control/command/turn_indicators_cmd`, `/control/command/gear_cmd`, and `/control/command/hazard_lights_cmd`.

Saves `steering_tire_rotation_rate` from `/control/command/control_cmd` in a new variable (`TargetSteeringTireRotationRate`).

Note: The following commands don't seem to exist in the newer autoware_msgs repository:
- `/control/current_gate_mode`
- `/control/command/emergency_cmd`
- `/control/command/actuation_cmd`

shen-kit, 2025-05-11:
Created a dummy autoware publisher (repo [here](https://github.com/Monash-Connected-Autonomous-Vehicle/autoware-dummy-publisher)), which can publish to `/control/command/turn_indicators_cmd`, `/control/command/gear_cmd`, and `/control/command/hazard_lights_cmd`. These have been tested manually and appear to work correctly.

dtonda8, 2025-05-11:
Great work! I like how you've designed the autoware dummy publisher to work with different topics. We'll test your SD-VehicleInterface branch with the actual vehicle (ENV200) during our next meeting. In the meantime, you can create a pull request for your VehicleInterface branch (but don't merge the PR just yet).